free-pdk/easy-pdk-programmer-software #3: PFS154 issues

spth, 2019-06-26:
I currently have two programmers, two PFS154 boards, two PFS173 boards (the f-eval boards from git - all identical except for PFS154 vs PFS173).
Using the current software, both programmers work fine with the PFS173 boards.

However, I see issues for the PFS154. Currently, only one programmer-board combination works sometimes (write sometimes works, sometimes fail at verify). The other 3 combinations always give 

~~~~
Erasing IC... done.
Writing IC... FPDK_ERROR: command ack failed / wrong icid
~~~~

I have not yet done further testing. I probably should try older software versions, as I remember both programmers working with both boards in the past. Also, I'll create a few more PFS154 boards and test on them.

spth, 2019-06-27:
I did a bit more testing. I used 5 programmers, 3 PFS173 boards, 7 PFS154 boards. All programmers work with all PFS173 boards.

For the PFS154, here is my compability matrix for this morning, trying at home:

~~~~
YNYYYYY
YYYYYYY
YYYYYYY
YNNYNYN
NNNYNYN
~~~~

Y means working, N means not working. We notice that there are 2 boards that work with all programmers and two programmers that work with all boards. Two of those marked N above would work about 1 out of 4 times. Some of those marked Y above didn't work when I tried them yesterday at my office at the unversity (there it was hotter, and the programmers were attached to a different USB hub). All results were obtained using easy-pdk-programmer.software compiled from source on a Debian GNU/Linux testing system.
Nicolas found similar results (using just two programmers and boards) on a Windows 10 system using the released binary, and without using an external USB hub.

Most of the time, the error was "Writing IC... FPDK_ERROR: command ack failed / wrong icid", but sometimes "FPDK_ERROR: verify failed".

Philipp

P.S.: Padauk recently updated their PFS154 and PFS173 datasheets, and made some amendmends in the section on programming. But I did not notice a difference between PFS173 and PFS154 there.

spth, 2019-06-27:
Is there anything else I could provide that could help in tracking down and fixing this issue?

freepdk, 2019-06-28:
Hi,

This sounds like the capacitor on VDD is causing problems (rise time of VDD needs to be different / VDD needs to be different, ...). 

In "development branch" (also requires new STM32 firmware compile + flash) I did some changes. Maybe this solves the problem.

Another solution could be to play with the voltages in "fpdkicdata.c" for PFS154 (e.g. lower VDD, increase VPP)

Here are my current values from development branch (already has some changes compared to master):
```
  { .name                         = "PFS154",
    .otpid                        = 0x2AA1,
    .id12bit                      = 0xAA1,          //OTP_WAY: 0x13
    .type                         = FPDK_IC_FLASH,
    .addressbits                  = 13,
    .codebits                     = 14,
    .codewords                    = 0x800,
    .ramsize                      = 0x80,
    .exclude_code_start           = 0x7E0,  //OTP area 16 words, contains empty space for user and BGTR IHRCR factory values
    .exclude_code_end             = 0x7F0,
    .vdd_cmd_read                 = 2.5,  //<==== 2.5 is lowest
    .vpp_cmd_read                 = 5.5,  //<==== maybe increase a bit 5.6  / 5.7  / 5.8  / ...
    .vdd_cmd_write                = 2.5,  //<==== 2.5 is lowest
    .vpp_cmd_write                = 5.5,   //<==== maybe increase a bit 5.6  / 5.7  / 5.8  / ...
    .vdd_write_hv                 = 5.8,  //<==== high voltage VDD for write (lower/increase this in case of verify errors)
    .vpp_write_hv                 = 8.5,  //<==== high voltage VPP for write (do not alter this to much, most likely unrelated)
    .write_block_size             = 4,
    .write_block_clock_groups     = 1,
    .write_block_clocks_per_group = 8,
    .vdd_cmd_erase                = 2.5,  //<==== 2.5 is lowest
    .vpp_cmd_erase                = 5.5,  //<==== maybe increase a bit 5.6  / 5.7  / 5.8  / ...
    .vdd_erase_hv                 = 3.0,  //<==== high voltage VDD for erase (lower/increase this in case of erase errors / not blank after erase)
    .vpp_erase_hv                 = 9.0,  //<==== high voltage VPP for erase (do not alter this to much, most likely unrelated)
    .erase_clocks                 = 2
  },
```

spth, 2019-06-28:
The 0.1µF capacitor is recommended by current Padauk documentation for PFS154 and PFS173, section 9.2.8, which also states that the capacitor should be below 0.22 µF.

freepdk, 2019-06-28:
> The 0.1µF capacitor is recommended by current Padauk documentation for PFS154 and PFS173, section 9.2.8, which also states that the capacitor should be below 0.22 µF.

Sure, it's in documentation. However the voltages / timings / ... are not mentioned. That's why we need to "play" with them. Original WRITER has a special option for in circuit programming which seems to modify voltages and timings. I just did not have the time to analyze this. I fixed the problems with the PFS154 board + 0.1µF cap with the new settings from above.

spth, 2019-07-01:
> In "development branch" (also requires new STM32 firmware compile + flash) I did some changes. Maybe this solves the problem.

Is the EASYPDKPROG.bin in the development branch sufficient? I tried to compile the firmware by commenting out the

`#GCC_PATH = ../../../../gcc-arm-none-eabi-8-2018-q4-major/bin`

in the Makefile, but get:

```
philipp@notebook5:~/free-pdk/easy-pdk-programmer-software/Firmware/source$ make
mkdir build		
arm-none-eabi-gcc -c -mcpu=cortex-m0 -mthumb   -DUSE_HAL_DRIVER -DSTM32F072xB -IInc -IDrivers/STM32F0xx_HAL_Driver/Inc -IDrivers/STM32F0xx_HAL_Driver/Inc/Legacy -IMiddlewares/ST/STM32_USB_Device_Library/Core/Inc -IMiddlewares/ST/STM32_USB_Device_Library/Class/CDC/Inc -IDrivers/CMSIS/Device/ST/STM32F0xx/Include -IDrivers/CMSIS/Include -IDrivers/CMSIS/Include -Os -Wall -fdata-sections -ffunction-sections -MMD -MP -MF"build/main.d" -Wa,-a,-ad,-alms=build/main.lst Src/main.c -o build/main.o
In file included from Drivers/CMSIS/Include/core_cm0.h:44:0,
                 from Drivers/CMSIS/Device/ST/STM32F0xx/Include/stm32f072xb.h:132,
                 from Drivers/CMSIS/Device/ST/STM32F0xx/Include/stm32f0xx.h:155,
                 from Drivers/STM32F0xx_HAL_Driver/Inc/stm32f0xx_hal_def.h:46,
                 from Drivers/STM32F0xx_HAL_Driver/Inc/stm32f0xx_hal_rcc.h:45,
                 from Inc/stm32f0xx_hal_conf.h:193,
                 from Drivers/STM32F0xx_HAL_Driver/Inc/stm32f0xx_hal.h:46,
                 from Inc/main.h:29,
                 from Src/main.c:21:
/usr/lib/gcc/arm-none-eabi/7.3.1/include/stdint.h:9:16: fatal error: stdint.h: No such file or directory
 # include_next <stdint.h>
                ^~~~~~~~~~
compilation terminated.
make: *** [Makefile:158: build/main.o] Fehler 1
```

It has been a few months since I compiled something for the STM32 on that system (it has the standard Debian testing packages for arm cross-development installed ), but have not run into such an issue before.

freepdk, 2019-08-15:
resolved in development branch

closed 2019-08-15 by freepdk