free-pdk/easy-pdk-programmer-software #5: Probe does not recognize PFS173

jacobjennings, 2019-07-05:
Thanks for sharing your design.

I assembled a board, loaded firmware, connected a PFS173 (SOIC16). Probe fails:

```
$ easypdkprog probe
Probing IC... Nothing found.
```

I captured the per-pin data with a logic analyzer. You can download my capture and view it with their software: https://www.saleae.com/downloads/

Smaller lower resolution capture:
https://www.dropbox.com/s/7ih8tvkrcb0lwat/easypdk-probe-PFS173-smaller.logicdata?dl=0

~1gb high resolution capture:
https://www.dropbox.com/s/qqwybo7lcyldlz8/easypdk-probe-PFS173-failure.logicdata?dl=0

Channels in order:
0: vdd
1: pa7
2: pa6
3: pa5
4: pa0
5: pa4
6: pa3

spth, 2019-07-05:
Seems odd. Both with the release and the development branch of the software, the PFS173 always worked fine for me (I currently have 4 PFS173 boards, 6 programmers, all combinations worked), even when there were problems with PFS154.

jacobjennings, 2019-07-05:
My assembly was pretty messy with a heat gun and iron, need to do one of those DIY toaster reflow ovens. I will try another PFS173, do some more testing, then maybe assemble another programmer and see if I have more luck

freepdk, 2019-07-05:
Hi,

please try the "simpletest" / "easypdkprogtest" program WITHOUT anything attached to the programmer. 

release: 
$make simpletest
$./simpletest
development: 
$make easypdkprogtest
$./easypdkprogtest

It will show basic communication of programmer works, generates 5.00V for VDD and VPP (PA5), measures the output and displays the result. If it shows both times around 5.00V then programmer assembly is most likely good.

Also: Do you connect the PFS173 directly or is it soldered to a board. If it is soldered on a board, you need to make sure capacitor on VDD is small and that output draining things (e.g. LEDs) might interfere with the programming pins of PFS173.

freepdk, 2019-08-04:
@jacobjennings : Have you solved the problem?

freepdk, 2019-08-15:
seems resolved.

closed 2019-08-15 by freepdk