free-pdk/easy-pdk-programmer-software #6: PFS232 and PFC232 support

spth, 2019-08-29:
Padauk just realeased the PFC232 and PFS232, their first Flash Multicore-µC. It would be good to have support for them.

freepdk, 2019-08-30:
Support can be added only, if a real physical device is available. It is not possible to add support for data sheet only CPUs.

spth, 2020-04-29:
I have not seen any mention of PFS232 at Papdauk in quite a while. I suspect it has ben cancelled in favour of only releasing PFC232.

spth, 2020-05-23:
For my work on SDCC and the f-eval boards, I consider this the third-most important open feature request. ICs will arrive soon.

kaweksl, 2020-08-20:
Any update on that ? I have got PFC232-S16 and i can't get it to work. programmer gives icid 0xDD4, settings form PFS154 don't work (couldn't erase chip)

freepdk, 2021-05-14:
Support added (see development branch). Since the IC needs 7V on VDD for writing the hardware mod on the programmer is required.

closed 2021-05-14 by freepdk