free-pdk/easy-pdk-programmer-software #7: PFS172 support

spth, 2019-08-29:
Padauk just released the PFS172, a pdk14 flash µC. It would be good to have support for it.

freepdk, 2019-08-30:
Support can be added only, if a real physical device is available. It is not possible to add support for data sheet only CPUs.

spth, 2019-09-02:
Historically, Padauk released datasheets and devices very close in time. So I would except availability of the device at distributors within few weeks.

spth, 2020-05-23:
For my work on SDCC and the f-eval boards, I consider this the most important open feature request. ICs are on the way.

spth, 2020-06-07:
I do not know how much is supposed to work already in the pfs17 branch, but:
I tried the branch with a PFS172 today. easypdkprog probe works for me. easypdkprog write says:

~~~~
Erasing IC... done.
FPDK_ERROR: chip is not blank
~~~~

I tried two different PFS172 chips.

freepdk, 2020-06-07:
Hi,

this feature branch is not complete yet since the normal programing method does not work (VDD requires >8V for erase/write to work reliable).

Now looking at the "limited voltage programing mode". Just takes some time.

freepdk, 2020-06-08:
You can try the PFS172 branch now. Read / write works perfectly. Erase is having some glitches which I fixed in a very crude way for now (erase works, but not like the original WRITER is doing it).

spth, 2020-06-09:
The pfs172 branch works for me. I tested with both PFS172 and PFS154. Thanks.

I've programmed the PFS172 about 10 times and so far did not see a "verify failed".

freepdk, 2020-06-09:
It happened to me when I tried to write a bigger program (>80% used).

spth, 2020-06-09:
I see. I tried only smaller programs meant for https://github.com/free-pdk/sdcc-pdk-code-examples so far.

cpldcpu, 2020-06-10:
What should be limiting is the number of zero bits that are written at a time. Maybe it helps to write nibbles instead of bytes? Or make this adaptive, so that a maximum of 4 zeros is written at any time?

freepdk, 2020-06-10:
Right now no limiting at all is implemented. Just the reduced voltage.

This works ok if you write only first 25% of IC.

So maybe just a small pause after every 4 word group might be enough.

I will do some experiments later.

freepdk, 2020-06-16:
Adding PFS172 support is finished and available in development branch right now

closed 2020-06-16 by freepdk