free-pdk/easy-pdk-programmer-software #8: PFS154 support

spth, 2019-08-29:
Padauk just released the PFC154, which seems quite similar to the already supported PFS154. The main difference seems improved ESD/EFT capabilities, and possibly the process used (0.5 µm vs 0.18 µm?).

freepdk, 2019-08-30:
Support can be added only, if a real physical device is available. It is not possible to add support for data sheet only CPUs.

spth, 2019-10-10:
A while ago, Padauk announced the PFC154. Yesterday, the datasheet was released.

I guess it will take a few more weeks until devices become available at distributors.

Section 13 on program writing in the PFC154 datsheet looks very similar to the corresponding section 9.2.8 in the PFS154 datasheet. A notable exception is the voltage on PA5 (5V on the PFC154 vs 8V on the PFS154).

freepdk, 2021-05-02:
PFC154 support is done. See development branch.

Unfortunately PFC154 wants 7.0V on VDD for writing which requires a hardware modification to the existing easypdk programmer. The resistor R6 needs to be change to 20k. Programmer firmware will auto detect the modification and adapt accordingly.

closed 2021-05-02 by freepdk