free-pdk/easy-pdk-programmer-software #11: Writing of last instruction word is incomplete

cpldcpu, 2019-09-01:
I came across a very interesting bug. It appears that the last instruction word is not completely written to the flash. See screenshot below showing a disassembly of the hex file before writing (right side) and after reading back from the device (PFS154, left side).

In this case a "ret" instruction is encoded here. Usually, when the source is only one file and the main loop never finished, this RET is never reached. However, when linking more than one source file, SDCC puts all subroutines there, so that the suddely becomes very important.

[image]

freepdk, 2019-09-02:
Hi,

Unfortunately it never happened to me. More information is needed.

Please provide a sample IHEX file and the exact easypdkprog commands you use to reproduce (WRITE , READ, ...)

cpldcpu, 2019-09-02:
Full example project attached. You can find the intel hex in "example/pfs1_xxx_blinky/main.ihx". It's a blinky on PA4. It does not work unless I create a dummy function at the end of the program as shown in "include/delay.c".

An exampe of a memory dump read back from the device is in "example/pfs1_xxx_blinky/out.bin". You can see that the last byte is 0x3f instead of 0x00.

The instructions used for programming are those from the makefile.
	$(PROGRAMMER) -n $(MCU) write $(TARGET)

[Toolchain.zip](https://github.com/free-pdk/easy-pdk-programmer-software/files/3564668/Toolchain.zip)

freepdk, 2019-09-02:
Thanks for the files... same happens here. Strange that I never saw this before.

Fixed in commit 7778a53e8ca460d33e9893f70c1100116e6dfcaa

closed 2019-09-02 by freepdk

cpldcpu, 2019-09-02:
Thank you for quick fixing! Will you generate a new release/exe?

freepdk, 2019-09-02:
I added a new development tag, so auto build will create a new release anytime soon.