free-pdk/easy-pdk-programmer-software #12: PMS15A support

spth, 2019-09-02:
A while ago, Padauk released the PMS15A. It would be good to have support for it.

While LCSC currently does not stock it (I guess no one asked them for it so far), other vendors supply it.

The PMS15A is the lowest-end device by Padauk so far. Since part of the popularity of these devices is their low price, I'd expect the PMS15A to become quite popular among the OTP devices.

spth, 2019-09-13:
Looks like the PMS15A has arrived (looks like one can get ten for a cent):
http://www.91taohuo.com/haohuo/detail/591310744735
https://detail.1688.com/offer/591310744735.html

freepdk, 2019-09-22:
Turns out, PMS15A is same IC as PMS150C (verified with real ICs).

==> PMS15A is a marketing derivate only. It is supported now by adding an alias to PMS150C.

closed 2019-09-22 by freepdk

Kashouryo, 2021-02-20:
Here's an update on this topic. Padauk's official writer will enforce the difference between PMS150C and PMS15A. The writer will not write PMS150C's binary into a PMS15A.

freepdk, 2021-02-20:
Hmmm...

I wonder how WRITER determines the IC type.

The PMS15A and the PMS150C I got both have exact the same readout except one value which is used for BGTR (band gap tuning).

PMS150C: @0x7EC : AA 01  (ret 0xAA)
PMS15A: @0x7EC: 7A 01 (ret 0x7A)

Can you make a readout of one of your empty PMS15A and attache it here? 
(You can attach ZIP files by drag&drop them to the comment here)

Maybe the PMS15A I bought are in fact PMS150C? Thanks PADAUK for not having any labels we can verify...