free-pdk/easy-pdk-programmer-software #14: IHRC trimming, use different pins on devices with 5 wire interface.

cpldcpu, 2019-09-22:
Right now the IHRC trimming routing is using PA4 on PFS154/173. PA4 is not part of the programming interface on these devices and requires running an extra connection to the test board. This could be simply fixed by using PA3 instead. No change possible for 6 wire interface, of course.

freepdk, 2019-09-22:
Hi,

not sure that this is possible since programmer uses SPI hardware to high speed count the clocks generated from MCU. Remapping the pin might cause problems capturing the high frequency.

However you reference the special in circuit programing wiring for PFS. I think for that case the in ROM factory calibration value for 16MHz is for.

cpldcpu, 2019-09-22:
Indeed, no SPI on the other pins... I will look into using the factory programmed values. This would reduce programming time anyways.

One thing that was a bit of a concern is that the programmer did not notice that PA4 was not connected and the calibration failed. Probably there were some spurious pulses on on PA4 when it was floating.

freepdk, 2020-06-16:
5 wire interface is for flash parts only, which all come with factory calibrated IHRC / BG value. So for the special programing case that only 5 wires are connected I think factory trimming values are good enough. If not, just connect one more wire... or use a socket for ic.

closed 2020-06-16 by freepdk