free-pdk/easy-pdk-programmer-software #18: fpdkicdata.c - misprint?

arsonoff, 2020-03-04:
1) OTP Program Memory: PMS150C - 1KW; PMS15A - 0.5KW
Memory size is different. They have an identical identifier?

2) OTP Program Memory: PMS133 - 3KW; PMS134 - 4KW
Similarly

3) PMC251 - 60 Bytes data RAM
.ramsize = 0x3C (not 0x60)

Question:
vpp_write_hv for OTP: PMS154B, PMS154C - 11V, for the others - 10.5V
It essentially? What possible deviation? These are volts or conditional coefficient?

spth, 2020-03-04:
1. I don't know of a reliable way to distinguish these two from the programmer. AFAIK, they are the same hardware both with 1 KW of memory, just different bins (i.e. I guess when there is a fault in the upper 0.5 KW of memory it gets binned as PMS15A, I guess some without a fault get binned the same depending on demand).
2. I don't know about that one, but I guess it is the same situation as 1.

arsonoff, 2020-03-05:
> They have an identical identifier?

Found (in the IDE) the same ones, though.

freepdk, 2020-03-09:
Hi,

1.)
I tested PMS150C and PMS15A real ICs... They both have 1KW usable program memory.
Just the original IDE has an option in header file to artificially limit the program size of the compiler when choosing PMS15A. But you could choose PMS150C and simply write it on PMS150C and everything will work fine.
So like I wrote before: PMS15A is a marketing derivate of PMS150C (same IC inside).

2) I tested PMS133 and PMS134 real ICs... and both have same identifier when reading them.
So my guess is that there is just PMS134 with 4kW program memory and again PMS133 is marketing derivate (see 1.).

3) This looks like a real mistake... will be corrected, thanks.
Question: vpp_write_hv for OTP: PMS154B, PMS154C - 11V, for the others - 10.5V
It essentially? What possible deviation? These are volts or conditional coefficient?

VPP_WRITE_HV = VPP write HighVoltage (2nd part of programing sequence after CMD phase).
This is an absolute voltage value.

All the values are experimental tested. Sometimes I needed to increase VPP_HV a bit to overcome problems of specific ICs. Original WRITER also uses different voltages for different ICs.

closed 2020-03-09 by arsonoff