free-pdk/easy-pdk-programmer-software #23: No Programmer Found

Gruftgrabbler, 2020-06-05:
Hey Folks! I tried to probe the Programmer after uploading the firmware to it, but without success. As I'm sitting on a macOS machine I don't know if it has anything to do with that, but I have 2 programmers here and easypdkprog can't probe neither of them. I successfully installed SDCC, uploaded the firmware to the programmer and compiled some Example Code for the PFS137, but unfortunately uploading the program doesn't seem easy. As you can see below I had success with installing the easypdkprog I guess, but I would be very thankful for any help I can get. Sitting on this problem for almost 2 weeks now and its becoming really frustrating.

```
dfu-util -d 0483:df11 -a "@Internal Flash  /0x08000000/064*0002Kg" --dfuse-address 0x08000000 -D EASYPDKPROG.dfu
```
I uploaded the firmware to the Programmer via DFU mode and got the information 'File download was successfully'.

```
(base) USERS-iMac:EASYPDKPROG USERNAME$ cp easypdkprog /usr/local/bin/
(base) USERS-iMac:EASYPDKPROG USERNAME$ easypdkprog -V
easypdkprog 1.2
(base) USERS-iMac:EASYPDKPROG USERNAME$ easypdkprog probe
No programmer found
(base) USERS-iMac:EASYPDKPROG USERNAME$ 
```

freepdk, 2020-06-07:
Hi,

when you get DFU mode working then also USB will work afterwards (no electrical problems).

In order to help  you please provide complete output from your DFU programming.

It sounds like you tried to downloaded the DFU file from github directly by using "RIGHT-CLICK" on the DFU file name in the repository... which in fact saved a HTML file. And programing a HTML file as firmware will obviously not work.

Suggestion: Download one of the RELEASES (ZIP file) containing the release version of everything.

Releases can be found on top bar in repository under the "Releases" button or use this link: https://github.com/free-pdk/easy-pdk-programmer-software/releases

Gruftgrabbler, 2020-06-08:
Maybe I wasn't clear enough to describe exactly what I already did. So I redo the DFU step in order to show what happens:

Just for the purpose of a full documentary. I used the "system_profiler SPUSBDataType" command on the terminal to list up all external USB devices, including the STM32Bootloader in DFU mode. I cut out most of the USB devices, because they're not important and will blow up this message a lot.

```
(base) USERS-iMac:Firmware USERNAME$ system_profiler SPUSBDataType
USB:

    USB 3.0 Bus:

      Host Controller Driver: AppleUSBXHCISPT
      PCI Device ID: 0xa2af 
      PCI Revision ID: 0x0000 
      PCI Vendor ID: 0x8086 

        USB2.0 Hub             :

          Product ID: 0x2812
          Vendor ID: 0x2109  (VIA Labs, Inc.)
          Version: b.e0
          Speed: Up to 480 Mb/sec
          Manufacturer: VIA Labs, Inc.         
          Location ID: 0x14500000 / 3
          Current Available (mA): 500
          Current Required (mA): 0
          Extra Operating Current (mA): 0

            STM32  BOOTLOADER:

              Product ID: 0xdf11
              Vendor ID: 0x0483  (STMicroelectronics)
              Version: 22.00
              Serial Number: FFFFFFFEFFFF
              Speed: Up to 12 Mb/sec
              Manufacturer: STMicroelectronics
              Location ID: 0x14530000 / 12
              Current Available (mA): 500
              Current Required (mA): 100
              Extra Operating Current (mA): 0

(base) USERS-iMac:Firmware USERNAME$ 
```
Now uploading the firmware to the programmer:
```
(base) USERS-iMac:Firmware USERNAME$ dfu-util -d 0483:df11 -a "@Internal Flash  /0x08000000/064*0002Kg" --dfuse-address 0x08000000 -D EASYPDKPROG.dfu
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Opening DFU capable USB device...
ID 0483:df11
Run-time device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuERROR, status = 10
dfuERROR, clearing status
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 2048
DfuSe interface name: "Internal Flash  "
Downloading to address = 0x08000000, size = 29068
Download	[=========================] 100%        29068 bytes
Download done.
File downloaded successfully
(base) USERS-iMac:Firmware USERNAME$ ls
EASYPDKPROG.dfu		LICENCE-ADDITONAL	README
(base) USERS-iMac:Firmware USERNAME$ 
```

And the devices list after uploading the firmware to the Programmer:

```
(base) Davids-iMac:Firmware davidwitulla$ system_profiler SPUSBDataType
USB:

    USB 3.0 Bus:

      Host Controller Driver: AppleUSBXHCISPT
      PCI Device ID: 0xa2af 
      PCI Revision ID: 0x0000 
      PCI Vendor ID: 0x8086 

        USB2.0 Hub             :

          Product ID: 0x2812
          Vendor ID: 0x2109  (VIA Labs, Inc.)
          Version: b.e0
          Speed: Up to 480 Mb/sec
          Manufacturer: VIA Labs, Inc.         
          Location ID: 0x14500000 / 3
          Current Available (mA): 500
          Current Required (mA): 0
          Extra Operating Current (mA): 0

            Easy PDK Programmer:

              Product ID: 0x5740
              Vendor ID: 0x0483  (STMicroelectronics)
              Version: 2.00
              Serial Number: 1234567855AA
              Speed: Up to 12 Mb/sec
              Manufacturer: free-pdk.github.io
              Location ID: 0x14530000 / 13
              Current Available (mA): 500
              Current Required (mA): 100
              Extra Operating Current (mA): 0

(base) USERS-iMac:Firmware USERNAME$ 
```

As you can see everything looks normal. The output of the DFU uploading process matches exactly the one given in the "Readme.md" in the EASYPDKPROG/Firmware Folder.
And after this process unplugging the programmer and replugging it will show up the device as "EASY PDK Programmer" in my devices List.

So I really don't know what I did wrong here and why I can't simply probe the device.
Thanks again for helping. I appreciate this a lot.

freepdk, 2020-06-08:
Hi,

Thanks for the output of DFU and from system profiler. It confirms that your programmer hardware is working fine (at least USB) and that enumeration took place.

Can you have a look in your "/dev" folder what kind of tty devices you get when programmer is connected?

`ls -al /dev/tty.*`

On a standard MacOS installation you should get the following:
```
crw-rw-rw-  1 root  wheel   18,   0 27 Mai 11:29 /dev/tty.Bluetooth-Incoming-Port
crw-rw-rw-  1 root  wheel   18, 0x00000152  8 Jun 11:15 /dev/tty.usbmodem1234567855AA1
```

If for some reason you do not see the "tty.usbmodem1234567855AA*" then usb serial on your mac might have problems. Have you installed 3rd party USB serial port drivers?

Gruftgrabbler, 2020-06-08:
Thanks again for this fast reply. I executed the commands and got both times the expected results. 
```
(base) USERS-iMac:/ USERNAME$ ls -al /dev/tty.*
crw-rw-rw-  1 root  wheel   18,   2 Jun  8 09:17 /dev/tty.Bluetooth-Incoming-Port
crw-rw-rw-  1 root  wheel   18,   0 Jun  8 09:17 /dev/tty.serial0
crw-rw-rw-  1 root  wheel   18,  12 Jun  8 13:04 /dev/tty.usbmodem1234567855AA
(base) USERS-iMac:/ USERNAME$ lsof | grep usbmodem
(base) USERS-iMac:/ USERNAME$ 
(base) USERS-iMac:/ USERNAME$ easypdkprog probe
No programmer found
(base) USERS-iMac:/ davidwitulla$
```

I wonder if there is any possibility that it has something to do with the easypdkprog executable.
I tried to execute it directly with ./easypdkprog and via copying it into the /usr/local/bin folder to be able to call if from everywhere

freepdk, 2020-06-08:
Hi,

for sure it has nothing to do with the location of "easypdkprog".

Your output from "ls -al /dev/tty.*" shows one device but for some reason the device number is not appended. It should read "/dev/tty.usbmodem1234567855AA1" for the first programmer.

This is very unusual for MacOS. What version of MacOS do you use?

I think you can get it to work easily by specifying the programmer directly as an option for easypdkprog (instead of using the auto search):

In your case (with missing "1" at the end):
```
./easypdkprog --port=/dev/tty.usbmodem1234567855AA probe
```

Gruftgrabbler, 2020-06-08:
YEAH! Thats it! :)
Your absolutely right specifying the port really does solve the problem:
```
(base) USERS-iMac:/ USERNAME$ easypdkprog --port=/dev/tty.usbmodem1234567855AA probe
Probing IC... found.
TYPE:FLASH RSP:0xEA2 VPP=4.50 VDD=2.00
IC is supported: PFS173 ICID:0xEA2
(base) USERS-iMac:/ USERNAME$ 
```
Im very thankful for your help. Now I can finally continue with my project and doesn't stuck at this stupid step anymore. I really appreciate your help 👍 

In fact I'm sitting on a 2018 Hackintosh machine, but normally everything works fine (expect some useless apple stuff like the remainder app) I can give you a system overview for clearance if you want:
```
(base) USERS-iMac:/ USERNAME$ sw_vers
ProductName:	Mac OS X
ProductVersion:	10.14.1
BuildVersion:	18B75
(base) USERS-iMac:/ USERNAME$ system_profiler SPHardwareDataType
Hardware:

    Hardware Overview:

      Model Name: iMac
      Model Identifier: iMac18,1
      Processor Name: Intel Core i5
      Processor Speed: 3,41 GHz
      Number of Processors: 1
      Total Number of Cores: 4
      L2 Cache (per Core): 256 KB
      L3 Cache: 6 MB
      Memory: 16 GB
      Boot ROM Version: IM181.0158.B00
      SMC Version (system): 2.39f6
      Serial Number (system): C02VDVKXH7JY
      Hardware UUID: 006D5EBE-533B-584B-92A6-70CD9573AB99
```
Again thank you!

closed 2020-06-08 by Gruftgrabbler

freepdk, 2020-06-08:
I'm not sure if this is related to hackintosh only or if this also can happen on normal Macs.

I changed the autodetect function to also try without the appended instance number. (In development branch ebd10b071fa52e808f5814c318d6fdbe01f35cf6).