free-pdk/easy-pdk-programmer-software #28: Display detected baud rate

serisman, 2020-06-28:
When using the 'start' command, it would be nice to see the baud rate that easypdkprog 'detected' from the MCU's autobaud transmission.

serisman, 2020-06-29:
I briefly looked at the code earlier today, and think this might be kind of hard to implement.

It looks like the programmer is making use of build in STM32 functionality to handle auto-baud, and after a brief glance at the capabilities it doesn't looks like one can read back the detected baud rate.  Maybe someone that is more familiar with the STM32 hardware can chime in on the feasibility.

Even if the baud rate could be read, it would have to be sent back to the software to be displayed which might complicate things further.

It would still be nice to have this feature, so I'll leave this reqest open until someone else can confirm the feasibility/difficulty.

freepdk, 2020-07-02:
Was not hard after all :-) See commit 6be0c9197770625039a210f2238adb8773abc36f

closed 2020-07-02 by freepdk

serisman, 2020-07-03:
> Was not hard after all :-) See commit [6be0c91](https://github.com/free-pdk/easy-pdk-programmer-software/commit/6be0c9197770625039a210f2238adb8773abc36f)

Excellent!  Thanks for knocking that out.