free-pdk/easy-pdk-programmer-software #38: Move includes and samples to separate repository

cpldcpu, 2020-07-01:
@serisman I like your refactoring of the includes very much. It's a nice way to retain compatibility to intellisense and keep everything organized.

My proposal would be to move the toolchain/includes, plus examples and potential library functions to a new repository within the free-pdk organization. Maybe "free-pdk-toolchain" ?

- Right now it is buried in the programmer firmware in a directory nobody will naturally look at.
- In addition, it would make a lot of sense to be able to prepare separate releases of toolchain and programmer.

We should also convert all the other examples (e.g. philips, mine) to the new include and clean everything up.

Another option would be to separate examples and toolchains into two repositories. 

@freepdk, @spth, @serisman  What do you think?

serisman, 2020-07-01:
Moving the Examples/Includes out of the programmer makes sense to me.

I don't think I would separate out the examples from the include files quite yet, but definitely if/when SDCC incorporates the pdk/ directory.

Maybe in the meanwhile we can create a new repo called "free-pdk-examples" (or even "easy-pdk-examples") that will have both examples and include files.  The appropriate include files can be removed once SDCC has them built in.

My only 'issue' with naming it "free-pdk-toolchain" is the potentially misleading term toolchain, which in reality includes SDCC and easy-pdk-programmer.

But, certainly open to thoughts from others.

serisman, 2020-07-01:
I created a placeholder repo (https://github.com/free-pdk/free-pdk-examples), but this can certain be renamed if needed.

I'll probably finish up my changes and try and get more feedback with my open pull branch / PR, but once closer to being done we can move it over to the new repo.

cpldcpu, 2020-07-01:
Great!  👍 

I agree on the naming - although I would still include a description how to set up the toolchain.

serisman, 2020-07-01:
> ... although I would still include a description how to set up the toolchain.

Yes, definitely.  I was figuring a README.md in the root with a section for how to set up the toolchain (among other things).

serisman, 2020-07-01:
FYI... I just finished double checking and making sure all registers for the PFS154 have valid BIT / MASK values, if anyone wanted to get in there and take a look before I do the same for the other devices.

https://github.com/free-pdk/easy-pdk-programmer-software/pull/33/files#diff-3cbdafe83e067ca7417c38a40ec96847

serisman, 2020-07-02:
https://github.com/free-pdk/easy-pdk-programmer-software/pull/33 is ready for review.  See the comment on the PR.

serisman, 2020-07-04:
> Moving the Examples/Includes out of the programmer makes sense to me.
>
> I don't think I would separate out the examples from the include files quite yet, but definitely if/when SDCC incorporates the [...]

I think I have changed my mind a bit.

I think we should have two new repos:
1. "free-pdk-includes" will contained all the common include files along with a root README.md that explains layout, intended usage, and includes a link to the "free-pdk-examples" repo.  This repo will be versionable with a root CHANGELOG.md.
2. "free-pdk-examples" will pull in the "free-pdk-includes" as a git submodule to an 'include' subdirectory, and will include a bunch of example programs, each in their own sub-directory with separate README.me and Makefile files.  There will be a root README.me that has links to, and explanations for the entire toolchain (i.e. SDCC, make, easy-pdk-programmer-software), as well as light information on how to setup/use the git submodule 'include' directory.

Having the includes broken out as a separate "free-pdk-includes" repo allows other repos to pull in new versions more easily (either manually, or through their own git submodule).  We could even leave some lightweight examples in the easy-pdk-programmer-software repo (i.e. the ones that are helpful to prove the programmer is working properly).

I'll work on getting this setup so we can see how it all works.

serisman, 2020-07-04:
Thinking about it a bit more, we might even want to split the includes up into two separate repos:
1. "pdk-includes" - the pdk/*.h files.
2. "easy-pdk-includes" - the easy-pdk/*.h files.

This allows them to be versioned separately, and makes it easier to pull into a larger project repo that has already has existing code in a root "includes" directory.  The "pdk-includes" repo could go away if/when SDCC directly incorporates them.

serisman, 2020-07-04:
I still have more work to do on the examples repo, but I proved the concept out:

- "pdk-includes" - https://github.com/free-pdk/pdk-includes
- "easy-pdk-includes" - https://github.com/free-pdk/easy-pdk-includes
- "free-pdk-examples" - https://github.com/free-pdk/free-pdk-examples

freepdk, 2020-07-05:
I had a look at the new repositories and it looks like it got really complex and 

- several git sub sub modules required in order to make a hello world 
- some features are hidden or not available anymore 
   e.g. how to tune to a specific target frequency like SYSCLOCK 8MHz TUNEFREQ 8123456 Hz (this is extremely useful when you try to match serial timings) 
- complex example code for hello world (useless setup() and loop() function)
- fixed magic values hidden inside of lib / submodule 
  e.g. serial_setup() is setting up what baud rate at what pins?
- no LICENSE information in project
- not mentioning original authors in "new" examples

serisman, 2020-07-05:
Thanks for reviewing.  I responded to your feedback on the PR.

freepdk, 2021-04-22:
new repo for includes exists

closed 2021-04-22 by freepdk