free-pdk/easy-pdk-programmer-software #39: Build failure on both regular MinGW and MSYS2 MinGW

Kashouryo, 2020-07-03:
**Describe the bug**
The software can not be build on both regular MinGW and MSYS2 MinGW

**To Reproduce**
mingw: make all
MSYS2 mingw: make all

**Expected behavior**
output exe

**Screenshots**
[image]
MSYS2 MinGW:
gcc -std=c99 -pedantic -Wall -O2 -Ilib/argp-standalone-1.3  -o easypdkprog easypdkprog.c serialcom.o fpdkutil.o fpdkcom.o fpdkicdata.o fpdkihex8.o fpdkiccalib.o fpdkicserial.o  lib/argp-standalone-1.3/libargp.a
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/10.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: fpdkutil.o:fpdkutil.c:(.rdata+0x0): multiple definition of `FPDK_ERR_MSG'; C:\msys64\tmp\ccIMxD7Q.o:easypdkprog.c:(.rdata+0xd00): first defined here
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/10.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: fpdkcom.o:fpdkcom.c:(.rdata+0x60): multiple definition of `FPDK_ERR_MSG'; C:\msys64\tmp\ccIMxD7Q.o:easypdkprog.c:(.rdata+0xd00): first defined here
collect2.exe: error: ld returned 1 exit status
make: *** [Makefile:31：easypdkprog] Error 1

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version 1903

freepdk, 2020-07-03:
This is very strange, since the auto build creates a mingw build everytime a commit is done on github.

See results here: 

https://travis-ci.org/github/free-pdk/easy-pdk-programmer-software/builds

The only difference I see with your setup is that you try to compile using the 64 bit MinGW.

freepdk, 2020-07-03:
From your output:
```
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/10.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: fpdkutil.o:fpdkutil.c:(.rdata+0x0): multiple definition of FPDK_ERR_MSG'; C:\msys64\tmp\ccIMxD7Q.o:easypdkprog.c:(.rdata+0xd00): first defined here C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/10.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: fpdkcom.o:fpdkcom.c:(.rdata+0x60): multiple definition of FPDK_ERR_MSG';

It looks like you compile using the mingw32 compiler running from ming64 environment. This is somehow a very strange combination.

freepdk, 2020-07-03:
I changed something in code which should fix this problem.

closed 2020-07-03 by freepdk