free-pdk/easy-pdk-programmer-software #40: Programming PFS173 fails on the development branch

cmfcmf, 2020-07-04:
**Describe the bug**

Programming a PFS173 does not work with the code from the development branch. 
It works fine from the master branch.

**To Reproduce**

- checkout master branch
- build easypdkprog, build firmware, flash firmware
- `easypdkprog --icname=PFS173 write Examples/helloworld_pfs173.ihx`
- `easypdkprog --runvdd=5.0 start`

-> works fine

- checkout development branch or #33
- build easypdkprog, build firmware, flash firmware
- `easypdkprog --icname=PFS173 write Examples/pre-compiled/PFS173/helloworld.ihx`

-> error:
```
Erasing IC... done.
Writing IC (184 words)... done.
FPDK_ERROR: verify failed
```

**Expected behavior**

Programming a PFS173 should work without errors.

**Desktop (please complete the following information):**
 - OS: Linux 64bit
 - Version: UBUNTU 20.04

**Additional context**

I also tested my own program compiled with SDCC and it has the same issue: It works with the master branch, but fails with the development branch or #33.

cmfcmf, 2020-07-04:
I `git bisect`ed through the history and reverting commit https://github.com/free-pdk/easy-pdk-programmer-software/commit/1c30d61eb373a71aece79e1ca884862bfaafb122 makes it work again.

I should also note that I am using the "Lite" programmer from here: https://www.eevblog.com/forum/blog/eevblog-1144-padauk-programmer-reverse-engineering/msg3106318/#msg3106318

serisman, 2020-07-05:
Hmmm, that's interesting.  It is working fine on my programmer.  I wonder if this is a case where the "lite" version is more sensitive to VPP/VDD than the original?

cpldcpu, 2020-07-05:
I haven't yet tried the development branch on the lite version. Will look into this.

The main difference between normal and lite version is that there is no power gating for the boost-converter. Differences could arise if the software relies on switching the voltages off via power gating instead of the DACs.

I looked at the diffs, but did not find any changes regarding that behavior between master and development branch. 

Need to set up the toolchain for the firmware first...

cpldcpu, 2020-07-05:
@cmfcmf  have you tried any chips other than pfs173?

cpldcpu, 2020-07-05:
Ok, I checked this now. Took a while to fix all the toolchains and versions, as I am using a combination of cygwin and wsl for the build. I just just use my linux box... well...

- PFS154 works perfective with the lite version
- PFS173 development branch fails on both "lite" and "normal" version of the programmer. So this is not a hardware issue.

cmfcmf, 2020-07-05:
> @cmfcmf have you tried any chips other than pfs173?

I only have PFS173 chips :)

freepdk, 2020-07-05:
This highly depends on the length of wires from programmer to IC, the connection of IC in socket (maybe some corrosion), wire / additional capacitance on VDD/PA.5, ...
Usually removing the IC from socket and inserting it again solves this kind of problems. 

On the other hand, when some of the "eval boards" appeared which do have an additional cap on VDD / something connected to PA5, it was needed to "play" with the voltages in order to make them work.

I just tested it with PFS173 in socket directly sitting on top of original programmer and everything is working fine. 

@cmfcmf Thanks for bisect.

Only the high voltage should affect the problem you are seeing. Change for PFS173 was:
.vpp_write_hv = 8.5,  =>  .vpp_write_hv = 9.0,

@cmfcmf @cpldcpu 

Please describe your hardware setup in full detail (e.g. 150mm jumper wire connected from original programmer to IC socket, holding IC) and try to find a value which is working for you, preferably near to 9.0 (try 8.9, 8.8, 8.7, 8.6).

cpldcpu, 2020-07-05:
> Please describe your hardware setup in full detail

I used a PFS173-SO in a DIP adapter plugged directly into the programmer. No other things (wires, caps) were connected. I will investigate a bit further.

cpldcpu, 2020-07-05:
I changed these lines to their old values:

```
    .vdd_cmd_write                = 4.0, // was 2.5
    .vpp_cmd_write                = 8.0, // was 5.5
```

Now it works perfectly on the lite and the standard version, even with cpp_write_hv=8.0V

freepdk, 2020-07-05:
@cpldcpu 

8V during command phase sounds totally wrong.

vdd_cmd difference to vpp_cmd should be at least 2V

Original write never uses more than 5 / 5.5V on VPP during command phase.

Maybe try to change vdd_write_hv (e.g. on PFS172  5.3V was a magic value).

cpldcpu, 2020-07-05:
But these are actually the exact values from the master branch:

https://github.com/free-pdk/easy-pdk-programmer-software/blob/3a557ccef00feb9ac3aa20c5984c158370f77cd7/fpdkicdata.c#L222

Confirms the bisect findings from @cmfcmf

cpldcpu, 2020-07-05:
4/8/8.5 works
4/7/8.5 fails
4/7/9.0 fails
4/7.5/8.5 works
3/7.0/8.5 works
2.5/6.5/8.5 works
2.5/6.0/8.5 works
2.5/5.5/8.5 fails

So it seems the issue is about the command voltage difference.

freepdk, 2020-07-05:
I'm confused.

above you said you used ".vdd_cmd_write = 4.0, .vpp_cmd_write = 8.0"

I say "vpp_cmd_write = 8V is to high"

then you say it is like in master branch but reference vdd_cmd_write ?

cpldcpu, 2020-07-05:
I just linked to the first line of the section in the master branch. (How do you do multi-line links)

It reads:

```
    .vdd_cmd_write                = 4.0,
    .vpp_cmd_write                = 8.0,
```

freepdk, 2020-07-05:
@cpldcpu  Thanks for the values.

VDD cmd = 2.0V is a bit low for STM32 to see them as high r(especially with longer wires). That's why I started with 2.5V

But why do you list only 3 values in your table?

Current development branch introduced 4 voltages:

```
    .vdd_cmd_write                = 2.5,
    .vpp_cmd_write                = 5.5,
    .vdd_write_hv                 = 5.8,
    .vpp_write_hv                 = 9.0,
```

cpldcpu, 2020-07-05:
i kept vdd_write_hv at 5.8V since this was not changed between master and development.

freepdk, 2020-07-05:
I checked with real writer again and it looks like on PFS173 indeed VPP_CMD_WRITE starts with a voltage difference of 4V.

So my best guess would be to try:

```
    .vdd_cmd_write                = 2.5,
    .vpp_cmd_write                = 6.5,
    .vdd_write_hv                 = 5.8,
    .vpp_write_hv                 = 9.0,
```

Not sure if we should lower vpp_write_hv to 8.5.

cpldcpu, 2020-07-05:
Yes, those voltages work fine.

changiung vpp_write_hv does not seem to have a very big impact. I can go fairly low (e.g. 6.5V) without an issue. I guess the problem was really the voltage difference for entering the programming mode.

freepdk, 2020-07-05:
@cmfcmf Please check if the fix also works for you.

cmfcmf, 2020-07-05:
> Please describe your hardware setup in full detail (e.g. 150mm jumper wire connected from original programmer to IC socket, holding IC) and try to find a value which is working for you, preferably near to 9.0 (try 8.9, 8.8, 8.7, 8.6).

I used the same setup as @cpldcpu:

> I used a PFS173-SO8 in a DIP adapter plugged directly into the programmer. No other things (wires, caps) were connected. I will investigate a bit further.

---

> @cmfcmf Please check if the fix also works for you.

Yes, it works for me! :tada:

closed 2020-07-05 by cmfcmf