free-pdk/easy-pdk-programmer-software #44: Time for new release?

serisman, 2020-07-13:
@freepdk - Just wondering what your thoughts are about cutting a new release version (v1.3?) of this repo?

It seems like some users are hitting a bit of a snag with the new calibration routines, where for whatever reason they are not realizing that they have to first pull down the development branch as well as re-compile both the Firmware as well as easypdkprog software.  An official release would potentially help a bit.  Or, are there other things you want to add, finish, clean up, or test further before a new release?

freepdk, 2020-07-13:
From my side a release is ok. I tested the new firmware and software with most of my test ICs.
Basically I was waiting for the open pull request from @serisman so we have a decision with examples inside of software. 

For future I think the "helloworld" should be included, but the rest might be moved outside. I already moved the sound demo to showcases.

So for now we could release 1.3 from current development.

freepdk, 2020-07-13:
1.3 release is out now.

closed 2020-07-13 by freepdk

serisman, 2020-07-13:
> From my side a release is ok. I tested the new firmware and software with most of my test ICs.

Ok, great!

> Basically I was waiting for the open pull request from @serisman so we have a decision with examples inside of software.

Thanks for the consideration.  I have been slowly building out the free-pdk-examples repo, but now that we have a separate place to work on that, no reason to rush removal from this repo (unless/until also having them here starts to confuse users more).

The pdk-includes is also in a somewhat stable state, but we still need to finalize some things regarding direction (i.e. modular peripheral approach, and factory calibration code).  Hopefully there would be minimal (or no) changes needed to the actual example code even if we change direction.  Again, no reason to rush this.  It has its own playground at this point.

> For future I think the "helloworld" should be included, but the rest might be moved outside. I already moved the sound demo to showcases.

That is your call ultimately.  But, I was thinking maybe the calib-and-fuse-demo.c would make more sense to leave in this repo since it probably does the best job of testing the programmer/device and is least like the other examples that are already part of the free-pdk-examples repo (or are planned for it).  The 'hello world' example seems more like the other examples in the free-pdk-examples repo and a similar example is already included there.  Another potential option would be to combine the calib-and-fuse-demo and hello world example into one demo that fully tests the programmer (i.e. tests calibration as well as serial communication).

> So for now we could release 1.3 from current development.

👍