free-pdk/easy-pdk-programmer-software #45: Include git rev/branch in version output

Bingo600, 2020-07-13:
**Is your feature request related to a problem? Please describe.**
I'm a git noob , and was just bitten by not being able to switch to the development branch.
I could not get my (Tim's) lite programmer to do a calibration , as it was not in master.
I spent 2 days debugging my soldering etc , learned a lot. But could not get calibration to work.

**Describe the solution you'd like**
I would like for the easypdkprog & firmware to include the git version in the revision printout.

Improvement for output of  easypdkprog -V and easypdkprog --verbose probe

$ git rev-parse HEAD
78983e5aee2ab4bc16628405a0da99ca290f9af3

Or
$ git log -1
commit 78983e5aee2ab4bc16628405a0da99ca290f9af3 (HEAD -> master, tag: 1.3, origin/master, origin/development, origin/HEAD)

Could you in the makefile run a:  git rev-parse HEAD , save it in a "GIT_VER"
Then insert a -DGIT_BUILD=$GIT_VER  , and use that GIT_BUILD as a string in the -V output ?

That could come in handy , when noobs like me don't know how to checkout development via git ?

easypdkprog -V
easypdkprog 1.2 - Git:78983e5aee2ab4bc16628405a0da99ca290f9af3

Same for programmer firmware

$ easypdkprog --verbose probe
Searching programmer... found: /dev/ttyACM3
FREE-PDK EASY PROG - Hardware:1.2 Firmware:1.3 Protocol:1.3 Git:78983e5aee2ab4bc16628405a0da99ca290f9af3
Probing IC... found.
TYPE:FLASH RSP:0x1AA1 VPP=4.50 VDD=2.00
IC is supported: PFS154 ICID:0xAA1

Maybe see thread here
https://www.eevblog.com/forum/blog/eevblog-1144-padauk-programmer-reverse-engineering/msg3134838/#msg3134838

serisman suggests : I would think the releases would still just use the release number, but branches could use a git commit or possibly the git branch name and git commit.

I'm not a git Guru , maybe there's a more elegant way of extracting what branch/git-nuber one uses when compiling.
But i'm sure i would have gotten som efficient help if had posted a : easypdkprog -V or easypdkprog --verbose probe - Showing i was on a totally wrong branch/level.

TIA
Bingo

**Describe alternatives you've considered**
Learning git :-)

**Additional context**
When i worked on Versalon (an opensource arm programmer) , it was super usefull to be able to get info about the specific version , that a user problem was related to.

freepdk, 2020-07-13:
I will correct the version number.

Confusion was coming from serisman releasing new style includes and examples which depended on development branch.

freepdk, 2020-07-13:
The git tag is added to easypdkprog (see development branch).

Current development branch reports as:

```
./easypdkprog -V
easypdkprog 1.3-2-gef65826
```

1.3 (last tag)
2 (2 commits after tag)
gef65826 (commit hash)

In case something is modified in the sources without a commit, then "- dirty" is appended:

```
./easypdkprog -V
easypdkprog 1.3-2-gef65826-dirty
```
 
Release compiles (the automatic build via travis) will use the tag only. Next release with tag 1.4 will report as:
```
./easypdkprog -V
easypdkprog 1.4
```

Bingo600, 2020-07-13:
Looks neat

I was toying with these , to get the branch included

git log -1 --decorate --oneline | cut -d \, -f 1 | sed 's/(//'

git log -1 --decorate --pretty=oneline | cut -d \, -f 1 | sed 's/(//'

Your output is nicer , but for a beginner i would not notice if this was build from master or devel

Any chance of getting this into the firmware version too , in time

/Bingo

freepdk, 2020-07-13:
@Bingo600 
Mentioning development in version is not needed at all since master only has tagged versions.
This means in case there is any output after the tag (e.g. 1.3-x-HHHHHHH) means it is from development.
You also use the commit hash to find the exact version on github (the hash is after the -g:  1.3-x-gHHHHHHH".

So from 
```
easypdkprog 1.3-2-gef65826-dirty`
```
you find `ef65826` as hash and build the following url:
```
https://github.com/free-pdk/easy-pdk-programmer-software/commit/ef65826
```
which shows you branch and commit.

I also will have a look how to add this to firmware. Right now firmware is not in travis auto build process so using the final tag is not easy here.

Bingo600, 2020-07-13:
I was primarily thinking about adding the same "tag" to the firmware makefile , for when building @home

kaweksl, 2020-07-13:
-v arg should print also programmer version if connected, also printing firmware branch could be useful.
Currently firmware version is printed when issuing for example
`easypdkprog --verbose probe`
but that is not obvious

Bingo600, 2020-07-13:
Could this work ?

`$ cat fw-vers-patch 
diff --git a/Firmware/source/Makefile b/Firmware/source/Makefile
index 89bfba2..329f97d 100644
--- a/Firmware/source/Makefile
+++ b/Firmware/source/Makefile
@@ -1,6 +1,8 @@
 #GCC_PATH = ../../../../gcc-arm-none-eabi-9-2019-q4-major/bin
 BUILD_DIR = build
 
+EPDKVER ?= $(shell git describe --dirty)
+
 TARGET = EASYPDKPROG
 
 # debug build?
@@ -125,7 +127,7 @@ endif
 # compile gcc flags
 ASFLAGS = $(MCU) $(AS_DEFS) $(AS_INCLUDES) $(OPT) -Wall -fdata-sections -ffunction-sections
 
-CFLAGS = $(MCU) $(C_DEFS) $(C_INCLUDES) $(OPT) -Wall -fdata-sections -ffunction-sections
+CFLAGS = $(MCU) $(C_DEFS) $(C_INCLUDES) $(OPT) -D__FPDKSW__=\"$(EPDKVER)\" -Wall -fdata-sections -ffunction-sections
 
 ifeq ($(DEBUG), 1)
 CFLAGS += -g -gdwarf-2
diff --git a/Firmware/source/Src/fpdk.h b/Firmware/source/Src/fpdk.h
index 4c2be8b..11af2b8 100644
--- a/Firmware/source/Src/fpdk.h
+++ b/Firmware/source/Src/fpdk.h
@@ -21,7 +21,9 @@ along with this program.  If not, see <http://www.gnu.org/licenses/>.
 #include <stdbool.h>
 #include <stdint.h>
 
-#define __FPDKSW__ "1.4"
+#ifndef __FPDKSW__
+#define __FPDKSW__ "unknown"
+#endif
 #define __FPDKHW__ "1.2"
 
 typedef enum FPDKHWVARIANT
`

freepdk, 2020-07-13:
@Bingo600 

The problem is not adding it to the Makefile. The problem is to add the correct tag to the autobuild process.

Right now when you create a tag on master the travis autobuild uses this tag to include it into the release.
Since the firmware is not built with the autobuild right now there is no way to add the tag.

Example:
In future new master version is pushed and new tag 1.4 is added.
Then travis autobuild just compiles easypdkprog and creates the release
=> problem: firmware does not have new tag. Any previous (by hand) compiled firmware will have the old tag since the new tag is the trigger for autobuild.

=> solution: I need to add the firmware build to travis. this requires to add the arm build tools to the travis VM build process...

Have a look at ".travis.yml" and ".travis-build.sh".

And in case you did not have noticed there is a "build passed" badge on github. When you click on it you see the last autobuild: https://travis-ci.org/github/free-pdk/easy-pdk-programmer-software
and if you click on "Build History" there you see all autobuilds done: https://travis-ci.org/github/free-pdk/easy-pdk-programmer-software

Bingo600, 2020-07-13:
I think this is "Over my head" ... As in i have never used travis
If you add a new version , i do a git pull , makes easypdkprog , and a new firmware on my local machine.

But that is not what you meant by the above , i'm sure.

/Bingo

freepdk, 2020-07-13:
@Bingo600 
You don't have to use travis... Travis is used to automatically build the releases (compiled versions of the project):
https://github.com/free-pdk/easy-pdk-programmer-software/releases

Many people do not want to compile easypdkprog (especially Windows users) so the releases are a convenient way for them.

Your workflow - checkout master, compile yourself would indeed work without any problem. Just for the autobuild I need to add the firmware build.

This is not a hard task, it just needs some time. I also need to find a way not to install the arm tools and compile the firmware 3 times for the 3 releases Linux/MacOS/Windows.

freepdk, 2020-07-13:
Firmware autobuild and git tag/version in programmer firmware is working now (see commit 6c348f81b5bd57d0a19a159316458de59c6f1421)

@kaweksl 
> -v arg should print also firmware/protocol version if connected, also printing firmware branch could be useful.
> Currently firmware version is printed when issuing for example
> `easypdkprog --verbose probe`
> but that is not obvious

-V prints the version of the host program. It is not intended to connect somewhere (USB) and get version numbers of remote systems.

Maybe need to add extra command like "test" or "version" (instead of "--verbose probe") ?

Bingo600, 2020-07-14:
@JS

Beautifull 

$ easypdkprog -V
easypdkprog 1.3-10-gc5fabd3

$ easypdkprog probe --verbose
Searching programmer... found: /dev/ttyACM3
FREE-PDK EASY PROG - HW:1.2 SW:1.3 PROTO:1.3 (1.3-10-gc5fabd3)
Probing IC... found.
TYPE:FLASH RSP:0x1AA1 VPP=4.50 VDD=2.00
IC is supported: PFS154 ICID:0xAA1

I would say this one can be closed with success - elegant

Who does that ??

closed 2020-07-15 by freepdk