free-pdk/easy-pdk-programmer-software #47: USB issue under linux on mini PC (ZOTAC Barebone ZBox-CI323NANO)

cpldcpu, 2020-07-26:
**Describe the bug**
I am encountering a strange bug: The programmer will disconnect after issuing a "start" command and is unresponsive after that until reconnecting. This errors seems to be specific to the communication that takes place after starting the program on the MCU, as I can see the the "blinky" firmware I loaded into is is active for a very short time (~100 ms)

I am suspecting that this is actually an issue with the USB driver on my set up. As the setup is a bit exotic (MXlinux, ZOTAC Barebone ZBox-CI323NANO), I will simply not use it for now. The same programmer works well on my normal set up.

I still wanted to share in case someone has an idea.

DMESG output below.

[    0.000000] microcode: microcode updated early to revision 0x368, date = 2019-04-23
[    0.000000] Linux version 4.19.0-6-amd64 (debian-kernel@lists.debian.org) (gcc version 8.3.0 (Debian 8.3.0-6)) #1 SMP Debian 4.19.67-2+deb10u2 (2019-11-11)
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-4.19.0-6-amd64 root=UUID=0d4e0af4-6bd6-4e33-b8fa-15ff7e598c7c ro quiet splash
[    0.000000] x86/fpu: x87 FPU will use FXSAVE
[    0.000000] BIOS-provided physical RAM map:
...

[  143.737547] xhci_hcd 0000:00:14.0: WARN Cannot submit Set TR Deq Ptr
[  143.737558] xhci_hcd 0000:00:14.0: A Set TR Deq Ptr command is pending.
[  143.739602] xhci_hcd 0000:00:14.0: WARN Cannot submit Set TR Deq Ptr
[  143.739613] xhci_hcd 0000:00:14.0: A Set TR Deq Ptr command is pending.
[  143.742010] xhci_hcd 0000:00:14.0: WARN Cannot submit Set TR Deq Ptr
[  143.742021] xhci_hcd 0000:00:14.0: A Set TR Deq Ptr command is pending.
[  143.744224] xhci_hcd 0000:00:14.0: WARN Cannot submit Set TR Deq Ptr
[  143.744234] xhci_hcd 0000:00:14.0: A Set TR Deq Ptr command is pending.
[  143.746492] xhci_hcd 0000:00:14.0: WARN Cannot submit Set TR Deq Ptr
[  143.746502] xhci_hcd 0000:00:14.0: A Set TR Deq Ptr command is pending.
[  143.748600] xhci_hcd 0000:00:14.0: WARN Cannot submit Set TR Deq Ptr
[  143.748610] xhci_hcd 0000:00:14.0: A Set TR Deq Ptr command is pending.
[  143.750904] xhci_hcd 0000:00:14.0: WARN Cannot submit Set TR Deq Ptr
[  143.750914] xhci_hcd 0000:00:14.0: A Set TR Deq Ptr command is pending.
[  143.751035] xhci_hcd 0000:00:14.0: WARN Event TRB for slot 13 ep 2 with no TDs queued?
**[  143.800772] usb 1-5.3.1.3: USB disconnect, device number 14**
[  143.803273] cdc_acm 1-5.3.1.3:1.0: failed to set dtr/rts
[  149.144391] usb 1-5.3.1.2: new full-speed USB device number 15 using xhci_hcd
[  149.247354] usb 1-5.3.1.2: New USB device found, idVendor=0483, idProduct=5740, bcdDevice= 2.00
[  149.247371] usb 1-5.3.1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  149.247382] usb 1-5.3.1.2: Product: Easy PDK Programmer
[  149.247391] usb 1-5.3.1.2: Manufacturer: free-pdk.github.io
[  149.247399] usb 1-5.3.1.2: SerialNumber: 1234567855AA
[  149.257225] cdc_acm 1-5.3.1.2:1.0: ttyACM0: USB ACM device

freepdk, 2020-07-29:
All it shows is a generic USB disconnect. 

Possible reasons:
- programmer + IC + LED + ... is drawing to much power when you "start" the program
- XHCI driver / XHCI chipset problem (a lot of warnings before the disconnect in your log)
  If you search with google it looks like the "Intel Celeron N3150" has some problems/quirks with the xhci USB chipset under linux. Here a bug report related to STM32 and your log output: https://bugzilla.kernel.org/show_bug.cgi?id=202541
  => simple solution: Put a USB2.0 HUB in the chain (this will prevent XHCI driver from being used).

cpldcpu, 2020-08-06:
Thanks for looking into this. 

I also considered the power issue, but then we are only looking at <5mA when running the MCU. It should be far from the USB limitations. A already tried a USB hub with separate power to no avail.

The XHCI hint is a good one. I need to look into this.

closed 2021-01-30 by cpldcpu