free-pdk/easy-pdk-programmer-software #48: Output information about programmer hardware

cpldcpu, 2020-08-06:
**Is your feature request related to a problem? Please describe.**
Right now the information encoded in `_hw_variant` is not used anywhere.

**Describe the solution you'd like**
I would suggest to output this information when using the command line tool, for example as part of the "-v" reply. 

Or any other idea how to use it?

freepdk, 2020-08-13:
Hardware variant is used in the firmware to use different ADC channels already.

Till now there is no need to forward this information to host software at the moment. Ideally it should not matter for the host software what hardware variant is used at all.

freepdk, 2021-05-02:
easypdkprog shows HwVariant and HwMod when executed with "-v".

closed 2021-05-02 by freepdk