free-pdk/easy-pdk-programmer-software #50: usbd_cdc_if.c buffer addresses

mnusa2, 2020-09-05:
While moving the sources to STM32CubeIDE I noticed somewhat doubtful code.
This may be for a purpose or an error.
I have not noticed any erroneous behaviour based on this.

USBD_CDC_SetTxBuffer() and USBD_CDC_SetRxBuffer() use the same buffer address UserRxBufferFS.

From usbd_cdc_if.c istarting line 150.
/* Private functions ---------------------------------------------------------*/
/**
  * @brief  Initializes the CDC media low layer over the FS USB IP
  * @retval USBD_OK if all operations are OK else USBD_FAIL
  */
static int8_t CDC_Init_FS(void)
{
  /* USER CODE BEGIN 3 */
  /* Set Application Buffers */
  USBD_CDC_SetTxBuffer(&hUsbDeviceFS, UserRxBufferFS, 0);
  USBD_CDC_SetRxBuffer(&hUsbDeviceFS, UserRxBufferFS);
  return (USBD_OK);
  /* USER CODE END 3 */
}

mnusa2, 2020-09-10:
Just to note, the file is for Firmware.

freepdk, 2020-09-10:
This is just during initialization. It sets a TX buffer of 0 length (so any pointer is fine here).

Later when there is something to send (in "CDC_Transmit_FS" you can see that the buffer supplied from caller is given to USBD_CDC_SetTxBuffer).

So all is fine.

closed 2020-09-10 by freepdk