free-pdk/easy-pdk-programmer-software #52: PFS122 support

spth, 2020-10-29:
A while ago, Padauk released the PFS 122.

Since this is a flash device, I'd expect it to be popular among hobbyists, and it would be good to have support for it.

kibix, 2022-12-06:
I am happy to pay a bounty and support testing if PFS122 support can be added

bonnom, 2022-12-14:
JS added PFS122 to the development branch last week! So you can start testing it out. 
It seems like the PFS122 is just a PFS173

kibix, 2022-12-14:
Yes, thanks for the hint ! 
I tested today with the previous version, selecting the PFS172 (that is also as what the chip identifies), and it programs well. The include files for SDCC are missing a register for the PFS122 (which has 16bit results from AD, compared to 8bit on other models), but that is, obviously, trivial to fix.

So i can confirm that even without a specific PFS122 setting, the chip can be correctly programmed by selecting PFS172.

Cheers

Johannes

bonnom, 2022-12-14:
I think the PFS172 may have already had a 12bit ADC.
For the PFS173 they discovered at least 11bits of the adc https://free-pdk.github.io/undocumented-features#11-bit-adc-pfs173.
If you have an PFS172 you can try to read the lower adc bits from the adress 0x23!