free-pdk/easy-pdk-programmer-software #53: PFS123 support

spth, 2020-10-29:
A while ago, Padauk released the PFS123.

Since this is a flash device, I'd expect it to be popular among hobbyists, and it would be good to have support for it.

amcnicoll, 2021-04-26:
+1. It's MTP, has a 12-bit ADC, and high(er) pin count packages. Its potential applications are definitely outgrowing Mini C.

bonnom, 2022-12-13:
I compared the datasheets between the PFS123 and PFS173. It seems like they the PFS123 is just a PFS173 with a 12bit-adc instead of an 8-bit one. I couldn't find any other difference! 
Even the ADC address is the same. There is just an extra ADC address at 0x23 for the lowest 4bits of the adc. The 0x23 address is not used in the PFS173.

I don't know how to add new IC's to SDCC and easy-pdk.
I think it needs to be added be added "fpdkicdata.c" but I don't know if board information needs to be added anywhere else.

spth, 2022-12-13:
On the SDCC side, nothing needs to be done. SDCC already should support all pdk13, pdk14, pdk15 devices. pdk16 is not yet supported by SDCC, though. And SDCC currently only supports using a single core, so for multicore devices, when using additional cores, those will have to run hand-written asm code.

bonnom, 2022-12-13:
I compared the PFS123 and PFS173 ".INC" files of the official padauk IDE. The INC files are completely the same, even the OTP_ID number is the same. I couldn't find if the id12bit in the INC files so I dont know if even that is the same.
I also found out that the PFS172 and PFS122 INC files are the same, with same OTP_ID number.

I guess we only need to know the id12bit number of the PFS122 and PFS123 to officially support them.

cjacker, 2023-03-17:
> I compared the PFS123 and PFS173 ".INC" files of the official padauk IDE. The INC files are completely the same, even the OTP_ID number is the same. I couldn't find if the id12bit in the INC files [...]

id12bit of PFS123 is also 0xEA2, same as PFS173.  easypdk probe it as PFS173 and everything works without any problem.