free-pdk/easy-pdk-programmer-software #58: Issue with pfs173 and compile source code. Please help me!

Vanbatai123, 2021-05-27:
I compiled the source code to generate "easypdkprog.exe" and "easypdkprogtest.exe" but these two always show "No programmer found" in case the release here work fine _"https://github.com/free-pdk/easy-pdk-programmer-software/releases/tag/1.3"_(can  detect MCU but can't verify the code: _"FPDK_ERROR: verify failed"_
I think i need to measure the voltage, but "easypdkprogtest.exe" do not work.
Please show me how to compile the source code or send me "easypdkprogtest.exe" that run by your side.
I really need your help. 
Thanks in advance

Vanbatai123, 2021-05-30:
After trying to make a new hardware, i saw it can program the pfs173 chip, but verify still show error. I ignore verify step and it work, Hope you can fix this issue.
_$ ./easypdkprog -n PFS173 --noverify write helloworld_pfs173.ihx
Erasing IC... done.
Writing IC (184 words)... done.
Calibrating IC
* IHRC SYSCLK=8000000Hz @ 4.00V ... calibration result: 7966812Hz (0x83)  done._

closed 2021-05-30 by Vanbatai123

benlovage, 2022-12-16:
I got a similar issue with programming a PMS150C, and the issue seems to have something to do with the USB port used. I was also getting a lot of `No programmer found` errors. My laptop's USB 2.0 port would write most of the code, without finishing, but my desktop computer's top USB port succeeded the first time, without any `No programmer found` errors.