free-pdk/easy-pdk-programmer-software #60: [PFC232] Verification failed during flashing

kaweksl, 2021-06-27:
**Describe the bug**
Getting verify failed error when flashing PFC232
And on latest development version calibration fails too.

```
./easypdkprog -n PFC232 write Examples/src/build/helloworld_2cores_pfc232.ihx -v
Searching programmer... found: /dev/ttyACM1
FREE-PDK EASY PROG - HW:1.2 SW:1.3 PROTO:1.5 (1.3-52-ge064f5d)
HWVAR:0 HWMOD:1
Erasing IC... done.
Blank check IC... done.
Writing IC (325 words)... done.
Verifiying IC... FPDK_ERROR: verify failed
```

```
./easypdkprog -n PFC232 write Examples/src/build/helloworld_2cores_pfc232.ihx -v --noverify
Searching programmer... found: /dev/ttyACM1
FREE-PDK EASY PROG - HW:1.2 SW:1.3 PROTO:1.5 (1.3-52-ge064f5d)
HWVAR:0 HWMOD:1
Erasing IC... done.
Blank check IC... done.
Writing IC (325 words)... done.
Calibrating IC
* IHRC SYSCLK=8000000Hz @ 5.00V ... calibration result: 0Hz (0x00)  out of range.
ERROR: Calibration failed
```

**Desktop (please complete the following information):**
- Ubuntu 20.04 x64
- SDCC : mcs51/z80/z180/r2k/r3ka/gbz80/tlcs90/ez80_z80/z80n/ds390/pic16/pic14/TININative/ds400/hc08/s08/stm8/pdk13/pdk14/pdk15 4.0.3 #11850 (Linux)
```
./easypdkprog probe -v
Searching programmer... found: /dev/ttyACM1
FREE-PDK EASY PROG - HW:1.2 SW:1.3 PROTO:1.5 (1.3-52-ge064f5d)
HWVAR:0 HWMOD:1
Probing IC... found.
TYPE:FLASH RSP:0x1BA8 VPP=4.50 VDD=2.00
IC is supported: PFC232 ICID:0xBA8
```

**Additional context**
I have also tried version with PROTO:1.4 (commit a5e3d29472bc41658f46588ef918998dff4aaabb), 
PFC232 calibrates with --noverify only on programmer with R6 mod
and example code works.

```
./easypdkprog -n PFC232 write Examples/src/build/helloworld_2cores_pfc232.ihx -v --noverify
Searching programmer... found: /dev/ttyACM1
FREE-PDK EASY PROG - HW:1.2 SW:1.3 PROTO:1.4 (1.3-46-ga5e3d29)
HWVAR:0 HWMOD:1
Erasing IC... done.
Blank check IC... done.
Writing IC (325 words)... done.
Calibrating IC
* IHRC SYSCLK=8000000Hz @ 5.00V ... calibration result: 8002134Hz (0x6E)  done. 
```

helloworld_2cores_pfc232 programmed on PROTO 1.4, HWMOD:1 runs ok on programmer with latest firmware.

freepdk, 2021-09-01:
I tried it with 2 variants of the programer (HWMOD:0 / HWMOD:1 = R6 mod) and in both cases I can erase/write/calibrate/run the exact same example without any issues.

From your output it looks like erase is working.

`--noverify`will not help you to program a chip. In case there is a verify error then the wrong content was read back after writing. Then almost 100% for sure the IC will not work (calibration needs a working program as well). 

It would be interesting to read back the content of the IC after you get `FPDK_ERROR: verify failed`

./easypdkprog -n PFC232 read readback_helloworld_2cores_pfc232.ihx

BTW: I saw your "helloworld.ihx" is slightly bigger than mine (maybe you use older SDCC). 

In order to see differences please attach your "helloworld_2cores_pfc232.ihx" and your "readback_helloworld_2cores_pfc232.ihx".

Skywalkerf34, 2021-09-28:
@freepdk 

Hello again, now with software issues
Got same problem with my fresh new programmer

software version is 1.3

d:\Users\FidL\Documents\eagle\easyPDKprog\easy-pdk-programmer-software-master\release1.3\EASYPDKPROG>easypdkprog --icname=PFS154  erase
Erasing IC... done.

d:\Users\FidL\Documents\eagle\easyPDKprog\easy-pdk-programmer-software-master\release1.3\EASYPDKPROG>easypdkprog probe
Probing IC... found.
TYPE:FLASH RSP:0x1AA1 VPP=4.50 VDD=2.00
IC is supported: PFS154 ICID:0xAA1

d:\Users\FidL\Documents\eagle\easyPDKprog\easy-pdk-programmer-software-master\release1.3\EASYPDKPROG>easypdkprog --icname=PFS154  write helloworld_pfs154.ihx
Erasing IC... done.
Writing IC (184 words)... done.
FPDK_ERROR: verify failed

same result using PFS173

d:\Users\FidL\Documents\eagle\easyPDKprog\easy-pdk-programmer-software-master\release1.3\EASYPDKPROG>easypdkprog --runvdd=4.0 start
Running IC (4.00V)... IC started, press [Esc] to stop.
=> top LED (3) is shinning but nothing on serial...

freepdk, 2021-09-28:
@Skywalkerf34 

If VERIFY fails, then the IC is not programed properly and of course it can not run.

Most likely your problem is caused by long wires between programmer and IC and/or bad connection from IC to socket / ...

Could you do the suggested read out after writing (see previous post)? Then attach hex file you tried to write and the read back hex file.

Skywalkerf34, 2021-09-28:
@freepdk 

look to show some instabilities with programmer and someting said "nothing found" again...
 
Here is the readback for PFS154

[readback_helloworld.zip](https://github.com/free-pdk/easy-pdk-programmer-software/files/7243152/readback_helloworld.zip)

freepdk, 2021-09-28:
Your readback shows a complete empty IC. Not a single bit has been written.

=> VDD / VPP supply to IC seems unstable (long wires, bad soldering, bad contact to IC in socket, ...)

Skywalkerf34, 2021-09-28:
@freepdk 
I have 2 programmer that was looking to discover the MCU yesterday
today only one 'find the chip
easypdkprogtest give
1st one :  Cd: 4.99   vpp: 4.88    (vref= 3.30) confirmed real on pin
2nd one : Cd: 4.91   vpp: 4.58    (vref= 3.30)

The  first one cannot discover the MCU anymore :-(
the second one give the result I already send but time to time see also not able to rediscover the mcu

Should I buy a new socket adapter?

freepdk, 2021-09-28:
VPP 4.58 seems a bit low, 4.88 is on the edge. Do you measure the values with the IC connected?

Skywalkerf34, 2021-09-28:
@freepdk 

No as long as it is said not drive test with mcu on the programmer..... Should I ?

freepdk, 2021-09-28:
@Skywalkerf34 
No you should not. Then like I said, VPP 4.58 seems a bit low, 4.88 is on the edge.

Time to measure some things on PCB.
=> start easypdkprog test and measure + write down all pins from opamp. I will do same later and we can compare.

Skywalkerf34, 2021-09-28:
@freepdk 

here is
programmer "1"  that cannot detect anymore
pin 1 = 4.99
pin 2  =2.51
pin 3  =2.51
pin 4  =- 2.76
pin 5 = 1.188
pin 6 = 1.18_
pin 7 = 5.01
pin 8 = 7.57

programmer "2" that detect but fail programming
pin 1 = 4.99
pin 2  =2.51
pin 3  =2.51
pin 4  =- 2.79
pin 5 = 1.185
pin 6 = 1.185
pin 7 = 4.62
pin 8 = 7.26

freepdk, 2021-09-28:
I will do some measurements (all over the board) later and provide info.

What I can immediately see: Voltage on pin 8 on both devices is much to low. This is the feed from the DC-DC booster and should have >13V.

=> check values of R1 / R5
=> check soldering of D1
=> check R24 / C13

Skywalkerf34, 2021-09-28:
@freepdk 
thevalue and soldering a correct, I check output voltage at U6 between D1 and C4 and the value here is only 
prog 1  8.91v
prog 2 8.46v

The coil I am using is maybe not good enough ? ( cannot get the one on BOM I use CD43 6.8UH

freepdk, 2021-09-28:
@Skywalkerf34 
Since it looks like you still have a hardware problem, let's continue in the hardware project repo:
https://github.com/free-pdk/easy-pdk-programmer-hardware/issues/23

kaweksl, 2022-12-21:
Hi, sorry for not responding but i was busy

So i had to create new environment, currently i'm using 
```
FREE-PDK EASY PROG - HW:1.2 SW:1.3 PROTO:1.5 (1.3-62-g38968a3)
HWVAR:0 HWMOD:0
```
i had to use commit 38968a32222e2d7c19f5b1be41db2a49a4437e3a as with newer versions i have issues (HWMOD not detecting, can't write ihx from exmaples ) . 

I still have same issue with verification.

I'm attaching zip with 3 files: readout after erasing chip, original ihx that has beed written and readout after writing original ihx 
[PFC232_verification_fail_files.zip](https://github.com/free-pdk/easy-pdk-programmer-software/files/10280876/PFC232_verification_fail_files.zip)
 
after writing example code with disabled verification, uC seems to work ( using easypdkprog start )

kaweksl, 2022-12-21:
UPDATE: Looks like i'm getting some corruption at writing.
After writing to chip i'm reading same think every time i read. but readout changes between writes, in different locations.

kaweksl, 2022-12-23:
Seams like i might actually got some bad chips as some flash without issues