free-pdk/easy-pdk-programmer-software #63: Adapting Firmware to STM32F042C6Tx

iDoka, 2022-08-17:
Due to chip shortage, especially apply to STM32 family would be great to have possibility to use functional and pin-to-pin substitution for **STM32F072C8Tx**. The most appropriate candidate is **STM32F042C6Tx** (with good availability in my area).

So,  fast look on schematic can't find any specific features that can have only F072, perhaps F042 also can be used at functional level - only rebuild FW for F042 required. 

Is it possible to use the STM32F042C6Tx instead of the STM32F072C8Tx?
If so, can I ask to build FW also for F042? 🙏🏻

LovelyA72, 2022-08-17:
The essential component that F042 lacks is the DAC module, which FreePDK programmer use to produce the correct flashing voltage.

cnlohr, 2022-08-17:
That shouldn't be an issue. the 042 has really versatile PWMs, Would just need to add an RC to the inputs of U3

cnlohr, 2022-08-17:
I've just done **a lot** of projects with the 042 and have found it to be very versatile, convenient, and more convenient to procure chip for various projects.  Always been a proponent of it.