free-pdk/easy-pdk-programmer-software #65: helloworld on PFS173 fails

CodeFetch, 2022-09-05:
When flashing the .ihx file provided in this repo, the helloworld example works,
but when I compile it from source code using the SDCC snapshot, it doesn't.
Actually I couldn't get any of the showcase projects to work, yet...

Here are the files created by SDCC: [helloworld_pfs173.zip](https://github.com/free-pdk/easy-pdk-programmer-software/files/9492177/helloworld_pfs173.zip)

 - OS: Linux 64bit
 - Version: Debian Testing
 - Compiler: SDCC 4.2.8 (Snapshot Sep 1 10:30)

CodeFetch, 2022-09-05:
It works with the 2 year old version of SDCC 4.0.4 found in this repository: https://github.com/jjflash65/Padauk-pfs154

freepdk, 2022-10-06:
Do you use the release (master branch) or the development branch?

I remember that I added support for newer SDCC in development branch: https://github.com/free-pdk/easy-pdk-programmer-software/tree/development