free-pdk/easy-pdk-programmer-software #69: PFS132 support

kibix, 2023-03-06:
Padauk has vastly simplified their range and discontinued a lot of their nice CPUs.

one type which as been added os the PFS132, which can be compared with PFS172 and PFS122 but has a full fledged 12bit ADC and multiplier.

I have a large stock of those chips and can provide for making the programmer work.

kibix, 2023-08-28:
Hey, i am putting up a bounty of 500$ for this to be solved. I have PFS132 in house and will ship to whoever wants to attack this.
Should not be THAT complicated if you are good with padauk reverse engineering ..

abebeos, 2023-08-30:
- https://github.com/bountysource/core/issues/1586

kibix, 2023-08-30:
ok - did not realise that. i personally guarantee for the bounty for the one solving the PFS132 issue. even if bountysource will not pay out, i will cover.

abebeos, 2023-08-30:
> ok - did not realise that.

Remember that you can ask paypal for a refund (this buyer protection thing)

spth, 2024-03-10:
>     * why not use the original programmer?

The original Padauk programmer is neither free nor cheap, but bulky instead. easypdkprog ic small, free and relatively cheap.

>     * Project here looks unmaintained

Not much activity recently, but I guess it could be revived.

>     * Budget seems low (effort high for just one chip)

Kind of. But once you implement support for one chip, the experience should help with adding support for more (there are open feature requests for support for many more chips).

>     * one needs the original programmer and software, the open-source programmer and the PFS132 (and possibly some other chips for reference).

Depending on location, others working on Padauk support could be willing to provide these.