free-pdk/easy-pdk-programmer-software #73: Issue with programming PFS154

Bug18, 2023-11-14:
Hello.

I'm currently working with PFS154 and have some issues while programming. Almost every time I try to probe it or program it I receive "not found" or "wrong icid" (I successfully programed it only 2 out of 100 times).

First I checked all components and especially C10 and C2 capacitors and even replaced them on one programmer board with 10nF (in datasheet max. capacitance between Vdd and GND) but it didn't help.

I looked in datasheet and it says that programing voltages should be 7.5 and 8 volts for Vdd and Vpp respectively.
When programming succeeded (with verbose option on) it displayed 2 and 4.4 volts for Vdd and Vpp respectively.

I tried changing it in "fpdkicdata.c" file and then compiling it to .exe but couldn't get it to work. I was getting some errors connected to sh (basically when running make command on Makefile it recognized my machine as Linux and not Windows).

It would be of the great help if you could provide me with command You used to compile it to .exe file and some theories why is programming failing.

If you need any screenshots or additional data please let me know.

Bug18, 2023-11-17:
Solved.

closed 2023-11-17 by Bug18

navan93, 2025-04-09:
I too am seeing something similar, can you please share what solved it for you ?