free-pdk/easy-pdk-programmer-software #81: PFS154-s08 serial helloworld

dw5304, 2024-09-12:
the default hello world example doesnt seem to send uart out over the ic's pin?
when i run the easypdkprog.exe  -n PFS154 start i can see it does say

Running IC (5.00V)... IC started, press [Esc] to stop.
Hello World!
Hello World!
Hello World!

steps to reproduce
download the release from the site on windows. 
easypdkprog.exe -n PFS154 erase
easypdkprog.exe -n PFS154 write ..\Examples\helloworld_pfs154.ihx
Erasing IC... done.
Writing IC... done.

easypdkprog.exe -n PFS154 start
Running IC (5.00V)... IC started, press [Esc] to stop.
Hello World!

expect to see uart "hellow world" on pin 2 on the oscilloscope using 115200 ,  data bites 8, parity check none, stop bit 2, at least thats what I believe it should be based on the src documentation? 
**Screenshots**

[image]

**Desktop (please complete the following information):**
 - OS: Windows 11 pro

cpldcpu, 2024-12-22:
Its a bit unclear: Is the timing incorrect or is there no output at all?

dw5304, 2024-12-22:
No output at all as far as I can. Tell?

closed 2026-06-07 by cpldcpu