free-pdk/easy-pdk-programmer-software #85: PFS122B support

spth, 2024-12-22:
While the PFS122B might not be a particularly interesting device, since it is apparently just a minor variant of the PFS122, that also should make it easy to support, since we4 already have PFS122 support.

bnister, 2025-02-01:
Judging by the programming protocol, internally PFS122B must be a very different beast from the older PFS172/122.
A branch of my fork can read and write, but it's too much of a hack to be considered mergeable at this time https://github.com/bnister/easy-pdk-programmer-software/commit/05b027c09a105e84343abfc813609f0e38a54c3d

cpldcpu, 2025-02-01:
In that case it would definitely make sense to merge your changes. Do you think you can clean them up and submit a pull request?

bnister, 2025-02-02:
Not currently. 9 bit command requires an extension to the communication protocol, and I'm not even sure how the software and the firmware negotiate the specifics. Probably a string version of sorts, major and/or minor? Also, my branch can't currently probe the chip due to `_FPDK_GetIDIC` being too simple-minded at the time.
And I'd like to see what other available but unsupported devices need first. My own short list includes the new PFC460 which is due to be shipped, PGS134 and PGS152 with EEPROM which are available as well, some high EFT OTP ones still produced and sold, and newer PFS132, PMS152E/G etc which don't seem to be available (yet?) so I'd need some community help with testing.
PFS122B can be either programmed with quite high of a VDD requiring the R6 mod or the bizarre multi-pass pulse train which is in my fork; both protocols can't be readily specified with the current FPDKICDATA. I suspect there's more to come for other types, too.

cpldcpu, 2026-06-04:
Something very odd: It seems that PMS120 and PFS122B are almost identical, except one is OTP and one MTP? They also share the same ID.

https://www.padauk.com.tw/en/product/show.aspx?num=245&kind=41
https://www.padauk.com.tw/en/product/show.aspx?num=185&kind=42

Edit: I think they are the same - also the programming information is identical.

cnlohr, 2026-09-28:
ah, sorry, I lost track of this project. I may or may not return to it, but, thank you for the feedback.