free-pdk/easy-pdk-programmer-software #86: PMS152G and PMS152E support

spth, 2024-12-22:
Apparently, Padauk has replaced the PMS152 supported by easypdprog by two new devices: the PMS152G and the PMS152E. The former has more I/O, but less PROM than the latter. I hope the existing PMS152 support makes it easier to also support these variants.

cpldcpu, 2024-12-22:
Does it really make sense to add all of these devices as individual issues? Maybe one issue with a priorization table?
I would assume that most people who are interested in this toolchain buy their devices at LCSC. So, supporting devices that are not obtainable there is probably lower priority.

Then, the question is who is going to add the support? JS seems to be inactive.

I would also like to point out that there are still issue with compatibility breaking in SDCC and incompatibilites to some examples here in the repo.

cpldcpu, 2024-12-22:
Just an idea: We could add a poll here: 

https://github.com/free-pdk/easy-pdk-programmer-software/discussions/new?category=polls

spth, 2024-12-22:
IMO, it is better to have individual issues for the devices. They can be closed individually. I think having all devices in one issue that issue would get overwhelmed, and soon, one couldn't make much sense anymore of which devices have been requested for what reason or haven't.

spth, 2024-12-22:
I hope to find some time to  look into the Padauk devices a bit again from the second half of August. Beside working on a basic pdk16 SDCC port, that could include looking into adding support for a few devices to easypdkprog.

cpldcpu, 2024-12-22:
I am a bit worried that all these issues are detracting from the actual issues. Because these are in fact feature requests. Is it ok if I move them to the discussions? People will think this project is abandoned is so many unresolved issues are stacked.

We could leave some issues open for devices that are requested more often, e.g. PFS132. Although that is one device I cannot find at LCSC.

spth, 2024-12-22:
Well, from their titles they're clearly feature requests. I don't think anyone will mistake them for bug reports. They don't have bug tags either.

ArneRossius, 2026-02-06:
I just bought a bunch of PMS152G, so I had a quick look at this. It seems that the PMS152G is very similar to the PMS152, I was able to get read and write working with minimal changes (I don't have the genuine Padauk writer so I can't verify that these are the correct voltages, but I was able to program all 100 controllers on the first attempt, including verify & frequency calibration).

The main change is "exclude_code_start", a read of a blank PMS152G had data starting at 0x4F4.

I'm not sure what the "otpid" should be, so I just put the value output by the "probe" command there.

I also tried the write voltages from PMS150G (as PMS150**C** and PMS152 use identical values too), but they didn't work at all (PMS152G remained completely blank after 5 writes), so it seems there's no similarity between PMS150G and PMS152G. There's no scrambling required for PMS152G either.

```
  { .name                                      = "PMS152G",
    .otpid                                     = 0x7570, // ???????
    .id12bit                                   = 0x757,
    .type                                      = FPDK_IC_OTP2_2,
    .addressbits                               = 12,
    .codebits                                  = 14,
    .codewords                                 = 0x500,
    .ramsize                                   = 0x50,
    .exclude_code_start                        = 0x4F4,
    .exclude_code_end                          = 0x500,
    .cmd_read                                  = 0x6,
    .vdd_cmd_read                              = 2.5,
    .vpp_cmd_read                              = 5.5,
    .vdd_read_hv                               = 2.5,
    .vpp_read_hv                               = 5.5,
    .cmd_write                                 = 0x7,
    .vdd_cmd_write                             = 4.3,
    .vpp_cmd_write                             = 8.0,
    .vdd_write_hv                              = 5.8,
    .vpp_write_hv                              = 10.5,
    .write_block_size                          = 2,
    .write_block_clock_groups                  = 1,
    .write_block_clock_group_lead_clocks       = 1,
    .write_block_clock_group_slow_clocks       = 8,
    .write_block_clock_group_slow_clock_hcycle = 30,
    .write_block_clock_group_trail_clocks      = 1,
  },
```